ARM-software/vscode-keil-studio-pack #1: Failed activate Arm Tools license

dahlbTrumpf, 2023-06-20:
Type: Bug Report

**Describe the bug**
- Desktop or Web: Desktop
- OS and Version: Win 10 19044.2965
- VS Code Version: 1.79.2
- Extension Version: 1.4.4
- We have a uVision 5 with Flex license server.
- VS Code Settings->@ext:Arm.keil-studio-pack Mdk License Server Url: I put here the URL from the Flex Server
- Start VS Code->Notifications: Failed to activate Arm tools license

**To Reproduce**
Steps to reproduce the behavior:
1. Start VS Code
2. Click on 'File'
3. Click on 'Preferences'
4. Click on 'Settings'
5. Write under 'Search Settings': '@ext:Arm.keil-studio-pack'
6. Write under 'Mdk License Server Url': Url of the Flex License Server
7. Close 'Settings' Window
8. Restart VS Code
9. Error message: Failed to activate Arm tools license

**Expected behavior**
No error message appear

wing-py, 2023-08-01:
Me too.

situation: 
when I use command: $.\armlm.exe activate -product KEMDK-COM0 -server https://mdk-preview.keil.arm.com
it just said so: License server 'https://flex13011.compliance.flexnetoperations.eu' could not be reached

mcgordonite, 2023-08-07:
Hi @wing-py, the license activation built into the extension specifically activates the MDK 6 Community Edition license. This is for users who do not already have a license and is restricted to non-commercial use. If you already have a flex license covering Arm Compiler, you shouldn't need to activate the license or modify the settings to build solutions with Arm Compiler. Please let us know if that doesn't work for you.

See more information in the documentation here: https://developer.arm.com/documentation/108029/0000/Activate-your-license-to-use-Arm-tools

mcgordonite, 2023-10-30:
Hi @wing-py, please reach out if you need more support, but for now I will close this issue.

closed 2023-10-30 by mcgordonite