ARM-software/vscode-keil-studio-pack #2: ARM Keil VSCode Extensions

wilpless56, 2023-07-16:
Type: Bug Report

**Describe the bug**

installed extensions as described in the user guide
imported blinky example for STM32Nucleo-f446re
opened project in Keil Studio for VS Code
when trying to compile the code (run/debug) VSCode complains that it cannot open the include files of line 17, 18 and 20 of main.c
board is properly recognized 

- Desktop or Web: desktop
- OS and Version: Windows 10
- VS Code Version: 1.79.2
- Extension Version: as of of July, 16th
- Target Device: STM32 Nucleo-f446re
- Other extensions you installed (and if the issue persists after disabling them):

mcgordonite, 2023-07-19:
Hi @wilpless56, thanks for the bug report. I believe this is an issue with our intellisense feature that we're currently working on. On my system, I see the lines you mention highlighted as problems, but when I run the build itself it succeeds and generates an AXF.

Can you confirm that the build succeeds on your system too? You can use the Build button in the CMSIS panel, or run the "CMSIS: Build" command from VS Code's command palette.

The fix for the intellisense issue should be released in the next week or so. For now, you should be able to work around it by toggling the Target Type selection in the CMSIS panel a couple of times (this generates the compile_commands.json needed to power the intellisense integration).

wilpless56, 2023-07-19:
Hello Matthew,

to me, the point is that the example zip file that Keil provides, doesn’t contain the required .h files.
My build does not succeed, instead I get
***@***.***

Thanks for the support,
Wilfried

mcgordonite, 2023-07-20:
Those header files are contributed in [CMSIS Packs](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/index.html) rather than coming with the example project. The build tool `cbuild` knows where the packs are installed so includes the relevant files as needed by the project. If you haven't configured a `CMSIS_PACK_ROOT` variable, the packs are stored in `AppData\Local\Arm\Packs` in your home directory.

jkrech, 2024-08-27:
Closing due to inactivity

closed 2024-08-27 by jkrech