ARM-software/vscode-keil-studio-pack #5: What is the correct location to report problems?

ReinhardKeil, 2023-07-25:
The VS Code version of MDK6 Community seems to have many places to report issues.  
What is the correct location? 
How can I find as a user this location?

mcgordonite, 2023-07-26:
The GitHub issues page for the extension in question is the best place to report an issue. If you're not sure which extension is affected, you can report an issue on this extension pack repository for triage.

There are several ways you are pointed to the issues pages:
- The VS Code Marketplace has an "issues" link:
        [image]
- The READMEs have a "submit feedback" section:
        [image]

When you start a new issue, I've updated the issue templates to hopefully make things clearer:
[image]

mcgordonite, 2023-10-30:
There is now a page dedicated to answering this question: https://www.keil.arm.com/support/. The extensions link here to help users decide.

closed 2023-10-30 by mcgordonite