ARM-software/vscode-keil-studio-pack #6: Software Components View Empty

FerdinandEn, 2023-08-07:
Win11
VS Code 1.81.0
Keil Studio Pack: 1.10.0

I wanted to run some expriments. vcpk seems ok.

I can compile the sample program Blinky_FRDM-K32L3A6.

However, I cannot see any modules/packs under the Software components view.

Can you tell me what settings I have done wrong?

[image]

mcgordonite, 2023-08-22:
Hi @FerdinandEn, sorry for the delay in getting back to you. We've made a few changes to that view to help debug situations like this, in version 1.5.0.

Can you try loading the Software components view again? You should see an error message with more details.

FerdinandEn, 2023-08-22:
Hi @mcgordonite 

I have updated the tools. I have attached the project.  I store my packs in C:\Keil_v5\ARM\Pack. I have set the environment variable CMSIS_PACK_ROOT  to C:\Keil_v5\ARM\Pack

[image]
[ttttt.zip](https://github.com/ARM-software/vscode-keil-studio-pack/files/12407212/ttttt.zip)

mcgordonite, 2023-08-22:
It looks like there's an issue with the definition of the board `NXP::FRDM-K32L3A6` in version 13 of the pack `NXP::FRDM-K32L3A6_BSP`. It incorrectly specifies the mounted device. There are two options:
1. Remove the `board` entry from your csolution file. It doesn't buy you much over the device anyway.
2. Update to version 16 of the NXP pack, which seems to have the correct device specification.

mcgordonite, 2023-08-22:
> If the HOME environment variable is set, vcpk does not install the tools in the .vcpk user directory.

It sounds like this needs a separate issue. Please create a new issue for it on https://github.com/ARM-software/vscode-environment-manager/issues: this is the repository specific for the Environment Manager extension's vcpkg integration. Could you also include further details, such as what you set the HOME variable to specifically?

FerdinandEn, 2023-08-23:
After the change to version 16 I can see the packages.  But to compile the source code must be adapted, because in version 16 some definitions are changed. If you update to version 17 you have the same problem again. The packages are not visible. Additionally the source code must be adapted, so that one can compile.

Do you have an example for STM32 (STM32F746ZGT6 -Nucleo). There I could also test the debugging.

mcgordonite, 2023-08-23:
> After the change to version 16 I can see the packages. But to compile the source code must be adapted, because in version 16 some definitions are changed. If you update to version 17 you have the [...]

Did you try removing the `board` entry instead?

> Do you have an example for STM32 (STM32F746ZGT6 -Nucleo). There I could also test the debugging.

It looks like there's a Blinky example you can try out: https://www.keil.arm.com/boards/stmicroelectronics-nucleo-f746zg-revb-fbabf0d/projects/. You'll need to convert it from µVision to Csolution format: right click on the uvprojx file in the explorer and select "Convert to csolution".

FerdinandEn, 2023-08-24:
After commenting out the entry, I was able to see the packages. Package Version 17.

>   target-types:
    - type: FRDM-K32L3A6
\#      board: NXP::FRDM-K32L3A6
      device: NXP::K32L3A60VPJ1A:cm4

I downloaded the specified STM project, was able to compile and debug it.

Is it possible at the moment to open a RTOS.view, as with the MDK?

mcgordonite, 2023-08-29:
Glad to hear that resolved your problem.

> Is it possible at the moment to open a RTOS.view, as with the MDK?

That is on the roadmap, but isn't currently available in VS Code.

closed 2023-08-29 by mcgordonite