ARM-software/vscode-keil-studio-pack #8: Compilation failure

CylAlon, 2023-12-22:
I had a problem building the project according to the official data. First, I used stm32cubemax to generate a keil project for stm32f103. Then convert the project using Keil Studio Pack in vscode. Finally, I click build and the following problem appears.
"
CMake Error at CMakeLists.txt:76 (project):
  The CMAKE_AS_ARM_COMPILER:

    armclang

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "AS_ARM" or the CMake cache entry CMAKE_AS_ARM_COMPILER to the
  full path to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
"
I've tried multiple chips over and over again with the same result.Finally, it is found that a line of code is added to "tmp\stm32f103Test\stm32f103Test\CMakeFiles\3.25.2\CMakeCCompiler.cmake" : "Set (CMAKE_AS_ARM_COMPILER" C: / Software/Keil_v5 / ARM/ARMCLANG/bin/ARMCLANG. Exe ") "will be able to compile successfully. However, if you click Build(clear), the file will be rebuilt to its previous state.

HalfSweet, 2023-12-30:
Yes, I encountered a similar error. I tried switching to the pre-release version, but the problem remains the same. 
Here are some logs of what went wrong:

```text
> cbuild d:\Air001Test\F030_MDK6\MDK-ARM\F030_MDK6.csolution.yml --update-rte --packs --schema --context +F030MDK6
info cbuild: Build Invocation 2.2.1 (C) 2023 Arm Ltd. and Contributors
warning csolution: device 'STM32F030F4Px' does not support 'trustzone: non-secure'
D:/Air001Test/F030_MDK6/MDK-ARM/F030_MDK6.cbuild-idx.yml - info csolution: file is already up-to-date
D:/Air001Test/F030_MDK6/MDK-ARM/F030_MDK6+F030MDK6.cbuild.yml - info csolution: file is already up-to-date
D:/Air001Test/F030_MDK6/MDK-ARM/F030_MDK6.cbuild-pack.yml - info csolution: file is already up-to-date
D:/Air001Test/F030_MDK6/MDK-ARM/F030_MDK6+F030MDK6.cprj - info csolution: file generated successfully
info cbuild: Processing 1 context(s)
info cbuild: Retrieve build information for context: "F030_MDK6+F030MDK6"
=========================================================
info cbuild: (1/1) Building context: "F030_MDK6+F030MDK6"
d:\Air001Test\F030_MDK6\MDK-ARM\F030_MDK6+F030MDK6.cprj validates

M650: Command completed successfully.

M652: Generated file for project build: 'D:/Air001Test/F030_MDK6/MDK-ARM/tmp/F030_MDK6/F030MDK6/CMakeLists.txt'
-- The C compiler identification is unknown
CMake Error at CMakeLists.txt:74 (project):
  The CMAKE_AS_ARM_COMPILER:

    armclang

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "AS_ARM" or the CMake cache entry CMAKE_AS_ARM_COMPILER to the
  full path to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:74 (project):
  The CMAKE_C_COMPILER:

    armclang

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "D:/Air001Test/F030_MDK6/MDK-ARM/tmp/F030_MDK6/F030MDK6/CMakeFiles/CMakeOutput.log".
See also "D:/Air001Test/F030_MDK6/MDK-ARM/tmp/F030_MDK6/F030MDK6/CMakeFiles/CMakeError.log".
error cbuild: error executing 'cmake' configuration
error cbuild: error building 'd:\Air001Test\F030_MDK6\MDK-ARM\F030_MDK6+F030MDK6.cprj'
error cbuild: exit status 1
Process failed with exit code 4294967295
Process failed with exit code 4294967295
```

About the software version:
OS: Windows11
Keil Studio Pack: 1.13.0
VSCode: 1.85.1

Finally, here's a project file that went wrong, and I thought maybe this would help solve the problem
[F030_MDK6.zip](https://github.com/ARM-software/vscode-keil-studio-pack/files/13798344/F030_MDK6.zip)

vescovd1, 2023-12-30:
Same issue here ... tried several example projects on several different target devices.
always get the armclang
  is not a full path and was not found in the PATH.

The project converter seems to populate two different compiler lines in the vcpkg-configuraton.json like:

	"requires": {
		"microsoft:cmake": "^3.25.2",
		"microsoft:ninja": "^1.10.2",
		"arm:compilers/arm/armclang": "^6.20.0",
		"arm:compilers/arm/armclang": "^6.21.0",
		"arm:debuggers/arm/armdbg": "^6.0.0",
		"arm:tools/open-cmsis-pack/cmsis-toolbox": "^2.0.0-0"
	}

which seems to be an error ... at least in the editor as indicated by underline squigglys but does not seem to effect package installer. The higher version package seems to install ok.

As an experment I changed to "6.20.0" and reactivated environment.
This time I get a different error:
    FAILED: CMakeFiles/cmTC_81ae4.dir/testCCompiler.c.obj 
    c:\Users\vescovd1\.vcpkg\artifacts\fd3d78d7\compilers.arm.armclang\6.20.0\bin\armclang.exe   -mcpu=Cortex-M4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -o CMakeFiles\cmTC_81ae4.dir\testCCompiler.c.obj -c C:\KeilProjectsVSC\cmsis-rtos-blinky-4e8a119b\tmp\Blinky\Debug\CMakeFiles\CMakeScratch\TryCompile-yaj5mv\testCCompiler.c
    armclang: error: Failed to check out a license.
    The license file could not be found. Check that ARMLMD_LICENSE_FILE is set correctly.
    armclang: note: 
ARMLMD_LICENSE_FILE is not set. You must set this to the path to your license.
    armclang: note: 
Information about this error is available at https://developer.arm.com/support/lic87/m1
     General licensing information is available at https://developer.arm.com/support/licensing/
     If you need further help, provide this complete error message to your supplier or to Arm via https://services.arm.com/support/s/contactsupport
     - ARMLMD_LICENSE_FILE: unset
     - LM_LICENSE_FILE: unset
     - ARM_PRODUCT_DEF: unset
     - ARM_PRODUCT_PATH: unset
     - ARM_TOOL_VARIANT: unset
     - Product location: c:\Users\vescovd1\.vcpkg\artifacts\fd3d78d7\compilers.arm.armclang\6.20.0\sw\mappings
     - Toolchain location: c:\Users\vescovd1\.vcpkg\artifacts\fd3d78d7\compilers.arm.armclang\6.20.0\bin
     - Selected tool variant: product
    armclang: note: 
 - Checkout feature: compiler5
     - Feature version: 5.0202302
     - Flex error code: -1
    Product: Arm Compiler for Embedded 6.20 Professional
    Component: Arm Compiler for Embedded 6.20
    Tool: armclang [5e9ad700]

Which seems to indicate it wants a license file.

mcgordonite, 2024-01-02:
Hi all, thanks for reporting this issue. We're investigating an issue with the Arm Compiler 6.21.0 package. While we investigate, you can work around this by using 6.20.0 instead.

Update your vcpkg-configuration.json file to include just this entry for armclang:
```json
"arm:compilers/arm/armclang": "6.20.0",
```
Note the caret `^` has been removed. The caret in`^6.20.0` means "any version greater than 6.20.0 but less than 7", so 6.21.0 would be picked up.

> The project converter seems to populate two different compiler lines in the vcpkg-configuraton.json

Thanks, we will also investigate this issue.

HalfSweet, 2024-01-02:
It's great. It works for me :)

mcgordonite, 2024-01-04:
The Arm Compiler 6.21.0 bundle has now been fixed. If you have already seen this problem, then the broken version will be cached on your machine in `$HOME/.vcpkg` (`%HOME%\.vcpkg` on Windows). The simplest way to force downloading the fixed bundle is:
1. Close VS Code.
2. Delete the .vcpkg directory in your home folder.
3. Open VS Code.

This will download the tools again.

mcgordonite, 2024-01-04:
@vescovd1 the license issue you are seeing is probably unrelated to the vcpkg issue seen by other users. If you are using Arm Compiler for non-commercial use cases, you can activate the license by running the `Keil Studio: Activate tools license` command from the command palette.

closed 2024-01-04 by mcgordonite