ARM-software/vscode-keil-studio-pack #10: Link in page https://www.keil.arm.com/packs/tle984x_dfp-infineon/versions/ is not working

GiorgioMon, 2024-01-18:
The link to download the version 1.1.1 is not working

https://itools.infineon.com/cmsis_packs/TLE984x/Infineon.TLE984x_DFP.1.1.1.pack

Could you please take a look?

[image]

mcgordonite, 2024-01-22:
Thanks for reporting this @GiorgioMon. We will investigate why this version is not available. Is there a reason you need that particular version? The later versions appear to be available.

jkrech, 2024-01-22:
Note, that the pack vendor (Infineon) is responsible for hosting all pack versions listed in the `<releases>` section of the package description file: https://itools.infineon.com/cmsis_packs/TLE984x/Infineon.TLE984x_DFP.pdsc
While version 1.1.1 is clearly listed: `<release version="1.1.0" date="2018-01-10">`, the corresponding url 
```
<url>/<vendor>.<name>.$version.pack
https://itools.infineon.com/cmsis_packs/TLE984x/Infineon.TLE984x_DFP.1.1.1.pack
```
does not return the pack file.

While we are trying to contact Infineon on this topic you may use the specified support contact to report the missing pack version as well.
```
  <supportContact>http://www.infineon.com/cms/en/corporate/company/contact/support/index.html</supportContact>
```