ARM-software/vscode-keil-studio-pack #12: May I ask how to add a new. c/. cpp file to the project targets? Please.

tungchiahui, 2024-01-28:
Type: Feature Request
May I ask how to add a new. c/. cpp file to the project targets?Please.
And How can I add a new. h file to the Include Path?Please.
Can I only complete the above operations through Keil MDK5?
If that's the case, I hope you can update the relevant features Please. Thank you to the development team!
I am a heavy user of Linux and I hope to completely get rid of Windows, as Keil MDK5 can only run on Windows.
Could you please help explain? Thank you. Do you have any plans to update?

mcgordonite, 2024-01-29:
Hi @tungchiahui, this extension pack uses the [CMSIS Toolbox](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/) to build your projects. Projects are defined in the Csolution format.

To add a c/cpp file to your project, edit the cproject YAML file and add the new file to the `groups` tree. For example:
```yaml
project:
  groups:
    - group: My Source Files
      files:
        - file: ./relative/path/to/mynewfile.c
```
See the docs for more information: https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#source-file-management

Use `add-path` to add directories to the include path for your project. For example:
```yaml
project:
  add-path: ./relative/path/to/include
```

The user-editable project files are a different paradigm from µVision, but they enable better integration with continuous integration systems, better cross-platform support and hopefully a better experience overall.

tungchiahui, 2024-01-29:
@mcgordonite Thank you for patiently explaining to me. Thank you very much!

closed 2024-01-29 by mcgordonite