ARM-software/vscode-keil-studio-pack #13: Failed to load solution contexts for xxxxx.csolution.yml

tungchiahui, 2024-01-29:
Type: Bug Report

Sorry to trouble you, when I tried to convert an MDK5 project on Linux, I encountered an error: Failed to load solution contexts for/media/tungchiahui/6A0A73946EE82411/UserFolder/XiaomiCloud/Work/Source/ST-Cube MX/42.Flash_ LED_ Linux/MDK-ARM/O.Template_ Linux.csolution.yml Source: Arm CMSIS csolution (Extension).

The content of the output is:/home/tongchiahui/. vscode/extensions/arm.cmsis solution-1.24.1-linux-x64/tools/uv2csolution/uv2csolution/media/tongchiahui/6A0A73946EE82411/UserFolder/XiaomiCloud/Work/Source/ST-Cube MX/42.Flash_ LED_ Linux/MDK-ARM/O.Template_ Linux.uvprojx
Created solution:/media/tongchiahui/6A0A73946EE82411/UserFolder/XiaomiCloud/Work/Source/ST Cube MX/42.Flash_ LED_ Linux/MDK-ARM/O.Template_ Linux.csolution.yml
With projects:
-/media/tongchiahui/6A0A73946EE82411/UserFolder/XiaomiCloud/Work/Source/ST Cube MX/42.Flash_ LED_ Linux/MDK-ARM/O.Template_ Linux.cproject.yml

My configuration file is located below. Could you please help me take a look? Thank you.
[config_file.zip](https://github.com/ARM-software/vscode-keil-studio-pack/files/14085167/config_file.zip)

I think my VCPKG environment should be fine and my packs should have been downloaded and installed successfully.
I tried to search for a solution on GitHub and ARM websites, but couldn't find it, so I came to trouble you and ask for a solution.

mcgordonite, 2024-01-30:
Hi @tungchiahui, I couldn't reproduce the "Failed to load solution contexts" error notification on my mac or windows systems. However, I see a problem in the csolution file: the cproject begins with `0.`, which is failing validation for the YAML schema.

Do you see any other symptoms in the UI? For example, is the target type visible in the Context view dropdown? Can you build the solution?

You could try checking the developer tools console for errors. Run the "Developer: Toggle Developer Tools" command from the VS Code command palette, then open the Console panel. Check for errors relating to loading the context.

tungchiahui, 2024-01-30:
@mcgordonite Okay, thank you very much. I have tried renaming the project so that it no longer begins with 0. This way, it can be converted normally.Thank you very much for your help. Wishing you a happy life!

closed 2024-01-30 by tungchiahui