ARM-software/vscode-keil-studio-pack #15: When I 当我想要用命令行执行"Keil Studio:Activate tools license"时，报了以下这个错误，请问这是为什么？

Nan-Tian, 2024-02-26:
[image]

federicobozzini, 2024-03-01:
Hi @Nan-Tian . This is a known issue in the Keil Studio Pack. To activate a license, just open your program and follow the prompt from the VScode guiding you to the license activation. As an example you might see a "No license found" message on the status bar (at the very bottom) which you can click.

Nan-Tian, 2024-03-06:
> Hi @Nan-Tian . This is a known issue in the Keil Studio Pack. To activate a license, just open your program and follow the prompt from the VScode guiding you to the license activation. As an example [...]

Thank you very much, but I found that this problem was solved when I changed the plugin to a pre-release version some time ago. I am not sure why this is, but I really appreciate your answer.

LJ-Conqueror, 2024-09-30:
@Nan-Tian I also meet the problem, when I want to activate the community license, it always tells me a failure tip like below.
[image]
Can you help me? Thanks.

jkrech, 2024-09-30:
When you click on the "No Arm License" you can open the license management GUI. Can you please send a screenshot of that window, please

[image]

LJ-Conqueror, 2024-09-30:
@jkrech Mine looks different, I have tried updating my internet connection, switching my browser DNS, and closing Windows firewall, but ALL doesn't work, and I can't visit the URL(https://mdk-preview.keil.arm.com) via Chrome
[image]

mcgordonite, 2024-09-30:
Hi @LJ-Conqueror, this looks like an issue with the Environment Manager extension, not the Keil Studio Pack. Please raise an issue here: https://github.com/ARM-software/vscode-environment-manager/issues. Include the content of the log file mentioned in the notification at "C:\Users\Vigor Wave\armml\logs\2024-09-30_11-57-02_2300.log". Thank you!