ARM-software/vscode-keil-studio-pack #16: ‘xxx.h' file not foundclang(pp_file_not_found)

tungchiahui, 2024-03-24:
Hello, after I updated the latest version of MDK6, it automatically helped me install the clangd plugin. However, no matter how I configure the configuration of the clangd plugin, my header file cannot be found by VScode. Could you please help solve this problem? Thank you.

I have configured settings. json as follows :
`{
    "clangd.fallbackFlags": [
        "-I../**"
    ]
}`

[image]

[image]

tungchiahui, 2024-03-24:
I solved it.

I modified settings.json as following :
`{
    "clangd.arguments": [
        "--compile-commands-dir=${workspaceFolder}/tmp/Template_Linux/TemplateLinux"
    ]
}`

[image]

tungchiahui, 2024-03-24:
> I solved it.
> 
> I modified settings.json as following : `{ "clangd.arguments": [ "--compile-commands-dir=${workspaceFolder}/tmp/Template_Linux/TemplateLinux" ] }`

I found that there are still some false positives, such as not being able to recognize<_ config_site>.
I think this is due to the lack of a defined __ ARMCOMPILER_VERSION. Can there be a solution to define __ ARMCOMPILER_VERSION in the environment?

`// -*- C++ -*-
// Modifications copyright (C) 2023 Arm Limited (or its affiliates). All rights reserved.
//===----------------------------------------------------------------------===//
//
// Part of the LLVM Project, under the Apache License v2.0 with LLVM Exceptions.
// See https://llvm.org/LICENSE.txt for license information.
// SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception
//
//===----------------------------------------------------------------------===//

#ifndef _LIBCPP___CONFIG
#define _LIBCPP___CONFIG

#if !defined(__ARMCOMPILER_VERSION)
#  include <__config_site>
#endif`

[image]

CareyJWilliams, 2024-03-25:
Hi @tungchiahui! Thank you very much for raising this issue.

> I found that there are still some false positives, such as not being able to find<_ config_site>.
I think this is due to the lack of a defined __ ARMCOMPILER_VERSION. Could there be a solution to define __ ARMCOMPILER_VERSION in the environment Please?

__ARMCOMPILER_VERSION should indeed be configured in your environment for you by the extension.

Please could I request some more information from you:

- What operating system are you running?
- Is the example you're using public? I'll try to reproduce on my end if you can tell me where it's from
- Please ensure your current context is configured for the AC6 compiler
- Please ensure you have a licensed AC6/armclang on your path (if you can build your project, this should be the case)

Please could you also examine your configuration files both for the configured project and globally.
You can find them via executing the following in the command palette:

```
>clangd.userConfig
>clangd.projectConfig
```

They should both contain 'AddFlags' entries with __ARMCOMPILER_VERSION present, but I suspect they might not in this case.

tungchiahui, 2024-03-25:
> Hi @tungchiahui! Thank you very much for raising this issue.
>
>
> __ARMCOMPILER_VERSION should indeed be configured in your environment for you by the extension.
>
> Please could I request some more [...]

@CareyJWilliams Hello, thank you very much for replying to my question and helping me solve it.

I am running Kubuntu22.04 (Ubuntu kde).

My project was generated through CubeMX, which generated the STM32F103C8T6 chip project.
I have also uploaded its template to GitHub, and the following is the GitHub repository link.
[https://github.com/tungchiahui/CubeMX_MDK5to6_Template](url)

I am very certain that my Armlang is working properly because I can compile it successfully.

Then I found the following code in the compile_commands. json file of cmake, which should also indicate that my compiler is working properly.

`{
  "directory": "/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/MDK-ARM/tmp/Template_Linux/TemplateLinux",
  "command": "/home/tungchiahui/.vcpkg/artifacts/2139c4c6/compilers.arm.armclang/6.21.0/bin/armclang --target=arm-arm-none-eabi -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Core/Inc -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Drivers/STM32F1xx_HAL_Driver/Inc -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Drivers/STM32F1xx_HAL_Driver/Inc/Legacy -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Drivers/CMSIS/Device/ST/STM32F1xx/Include -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Drivers/CMSIS/Include -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Middlewares/Third_Party/FreeRTOS/Source/include -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Middlewares/Third_Party/FreeRTOS/Source/portable/RVDS/ARM_CM3 -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/applications/Inc -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/bsp/boards/Inc -I/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/MDK-ARM/RTE/_TemplateLinux -I/home/tungchiahui/.cache/arm/packs/ARM/CMSIS/5.9.0/CMSIS/Core/Include -I/home/tungchiahui/.cache/arm/packs/Keil/STM32F1xx_DFP/2.4.1/Device/Include -mcpu=Cortex-M3 -mfpu=none -mlittle-endian -DSTM32F10X_MD -D_RTE_ -DUSE_HAL_DRIVER -DSTM32F103xB  -g -O3   --target=arm-arm-none-eabi -c -isystem \"/home/tungchiahui/.vcpkg/artifacts/2139c4c6/compilers.arm.armclang/6.21.0/include\"  -std=c99 -fno-rtti -funsigned-char -fshort-enums -fshort-wchar -ffunction-sections -Wno-packed -Wno-missing-variable-declarations -Wno-missing-prototypes -Wno-missing-noreturn -Wno-sign-conversion -Wno-nonportable-include-path -Wno-reserved-id-macro -Wno-unused-macros -Wno-documentation-unknown-command -Wno-documentation -Wno-license-management -Wno-parentheses-equality -o CMakeFiles/Template_Linux.dir/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Core/Src/main.o -c /home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Core/Src/main.c",
  "file": "/home/tungchiahui/UserFolder/Sources/STM32F103C8T6/Core/Src/main.c"
},`

Thank you very much for your help.

CareyJWilliams, 2024-03-25:
What file did you find this original code snippet in, or have you added it yourself?
I can't see it within the example you've shared.

Eplankton, 2024-04-15:
Same problem

Eplankton, 2024-04-15:
> Hi again @tungchiahui, thank you for the extra info.
> 
> I cloned the repository above and tried opening the MDK-ARM folder inside STM32F103C8T6. I can confirm that the clangd project config file doesn't seem to be set right, as you mentioned initially. However, once I corrected that, I seemed to get the behaviour I'd expect, with both my clangd configuration files being populated with AddFlags including __ARMCOMPILR_VERSION etc.
> 
> If you can't figure out how to access your clangd config files via the instructions above you can also check for the presence of the define by hovering over it, like so:
> <img alt="Screenshot 2024-03-25 at 17 12 25" width="495" src="https://private-user-images.githubusercontent.com/3001447/316591977-e5ee4936-abca-44ad-a393-538d777c7f37.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxNzA0ODYsIm5iZiI6MTcxMzE3MDE4NiwicGF0aCI6Ii8zMDAxNDQ3LzMxNjU5MTk3Ny1lNWVlNDkzNi1hYmNhLTQ0YWQtYTM5My01MzhkNzc3YzdmMzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTVUMDgzNjI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Mjc1YmU4MjMxOGEwNzU5ODkxNjIzNTcxZjJhNDZmZDQyNzlkMDc0OGQ3NzBlYWNjMjQzZmJiN2VkMzk5NmJkYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.A8XJEe74N872vU7SbCstk_9h4BeZvCEzQeZMRSHBBqs">
> 
> You need to make sure the `cmsis-csolution.clangdArmclangMacroQuery` setting is enabled in order to get the filled in compiler defines (but this should be on by default).

Would you like to share the context of that clangd? which filled with flags like `__ARM_COMPILER_VERSION`

jmjoy, 2024-08-26:
I manually modified the `cproject.yml` file to add the include path (not configurable in the graphical interface), then deleted `.clangd`, closed the workspace and reopened it, and clangd worked fine.

https://github.com/jmjoy/study-stm32/blob/934be47cccfa809f689cebf9de96819340c7935d/001-LED/001-LED.cproject.yml#L3-L7

closed 2025-09-07 by jkrech