ARM-software/vscode-keil-studio-pack #17: Solution pack cannot be installed

15060501338, 2024-04-11:
During the installation, the software package required by the solution is being installed. After a while, the message "cannot to resolve packs STM32F1xx_DFP, CMSIS" is displayed.

[image]

15060501338, 2024-04-11:
[image]
It has always been 0kb, could you please install it manually? I can download the pack on kail's official website

jkrech, 2024-04-11:
As a work around could you open a terminal window in VSCode and run the installation commands manually?
`cpackget rm --purge Keil::STM32F1xx_DFP`
followed by 
`cpackget add Keil::STM32F1xx_DFP@2.4.1`
and
`cpackget add ARM::CMSIS@5.9.0`

You should see an output very similar to this:
```
cpackget add Keil::STM32F1xx_DFP
I: Adding pack "Keil::STM32F1xx_DFP"
I: Downloading Keil.STM32F1xx_DFP.2.4.1.pack...
I: 100% |███████████████████████████████████████████████████████████████████████████████| (50/50 MB, 3.0 MB/s)

I: Extracting files to C:\Users\xuser\AppData\Local\Arm\Packs\Keil\STM32F1xx_DFP\2.4.1...
I: 100% |█████████████████████████████████████████████████████████████████████████████████| (904/904, 658 it/s)

C:\workspace\cpackget_test>cpackget add Keil::STM32F1xx_DFP
```

If you see different output and errors, please report here.

15060501338, 2024-04-11:
Do you want to execute these three sentences in sequence

15060501338, 2024-04-11:
Can I download pack manually using a browser and install it from the command line

15060501338, 2024-04-11:
上述问题已经接近感谢作者

[image]
There's a new promotion over here. Do you need to fix this

Unable to load schema from 'https://raw.githubusercontent.com/Open-CMSIS-Pack/devtools/schemas/projmgr/2.3.0/tools/projmgr/schemas/csolution.schema.json': Request vscode/content failed unexpectedly without providing any details.

Finally, when will the device adaptation of jlink be added

Thank you for your patience. Thank you

jkrech, 2024-04-11:
> Do you want to execute these three sentences in sequence
Yes,  you need to enter these commands in a sequence, but you have solved the installation already. So you can ignore.

LJ-Conqueror, 2024-09-21:
> > Do you want to execute these three sentences in sequence
> > Yes,  you need to enter these commands in a sequence, but you have solved the installation already. So you can ignore.

@jkrech Can you help me? When I execute the pack add command, the information is displayed below.

read tcp 10.10.40.62:63908->72.21.81.200:443: wsarecv: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
E: failed writing HTTP stream to local file

jkrech, 2024-09-23:
I am afraid you have a network problem that is unrelated to the tools.
Can you please try to use your browser to download the packs? You can use the search box to locate a pack.
e.g. STM32F1: https://www.keil.arm.com/packs/stm32f1xx_dfp-keil/boards/
After download you can use `cpackget add /my/download/pack/Keil.STM32F1xx_DFP.2.4.1.pack` to install the already downloaded pack.

LJ-Conqueror, 2024-09-30:
@jkrech  I have solved my problem via your method, thanks.