ARM-software/vscode-keil-studio-pack #18: Licensing VCS Keil Studio for commercial use.

vescovd1, 2024-04-17:
VSC on Windows 10.

See Issue #11
[https://github.com/ARM-software/vscode-keil-studio-pack/issues/11](url)

Here it was stated that "Going forward you will get a User Based License (UBL) for MDK Professional which would enable both uVision and MDK6 VSCode."

We presently have a uVision, 10 user floating commercial licenses.
We have our license server on a mapped network drive.

How do we transition to use VSC Keil Studio for commercial use for our team?

HansSchneebauer, 2024-04-19:
Please raise a support request for this kind of question (see: https://developer.arm.com/All%20Support%20Services) and refer to both of your GitHub issues. Discussing licensing-related questions usually requires the exchange of confidential information.

kayasemih, 2024-06-04:
@vescovd1 We had the same problem. For our team, we use Keil uvision5 IDE with mdk 5.38 on a professional license. We wanted to use the vscode arm Keil studio extension to test a more practical and updated version. 

The result is a disaster. We couldn't integrate our old projects compiled with ARM Compiler v5.06 here. As if that wasn't enough, we were not allowed to use the existing commercial professional keys here in the Arm License activation step. When we wanted to test with Keil MDK Community (non-commercial free of charge), this time Keil uvision5 IDE exploded our existing licenses. 

I really have a hard time understanding ARM Keil, which is poorly supported, community support is always delayed, so weak, and not user-friendly. As an embedded developer, it is really ridiculous to deal with such environmental problems at the point we have reached technologically today. Please don't think that I am writing this message emotionally.  This is the output I got after my experiences and many trials. Don't you think that not allowing such a sharp transition just for commercial concerns does not arouse an unnecessary antipathy towards Arm products? 

Alternatively, with the https://em-ide.com/ platform, which also has a free licensed VS extension, we were able to use it easily despite the lack of a detailed user guide. We were also able to integrate old projects. Congratulations to the team who contributed.