ARM-software/vscode-keil-studio-pack #22: Compilation error when using the wrong compiler path

NguyenTrongPhuc552003, 2024-05-28:
[image]
### **_Type: Compilation path Issue_**

Hello, I am a user experiencing your ARM Keil Studio extension package. I got the compile error problem as shown in the picture (left: success, right: failure). When I create a new project, only Blinky compiles and debugs successfully, the rest all fail with the same error as in the image (on the right, probably the compiler path not found error).
- Board: FRDM-K32L3A6
- OS: Windows 11
- Keil MDK Community: non-commercial free of charge

Can you review and correct it for me? Thank you for reading !

mcgordonite, 2024-05-28:
Hi @NguyenTrongPhuc552003, is the compiler referenced in the vcpkg-configuration.json file for the failing projects? This is read by the [Environment Manager](https://github.com/ARM-software/vscode-environment-manager) extension to install the tools. You can see which tools are installed in the status bar ("Arm Tools: N"). In your left hand screenshot, 5 tools are installed. On the right, only 4.

NguyenTrongPhuc552003, 2024-05-28:
Hi @mcgordonite , as you said, most projects that fail to compile only have 4 tools. So I compared the successful compilation project and installed a 5th tool, details are as follows _(The remaining tools are created by default by the system with the included version)_:

- Arm CMSIS-Toolbox: 2.4.0
- Arm Compiler for Embedded: 6.22.0
- Arm Debugger: 6.1.2 _**(The 5th tool has just been added)**_
- Kitware's cmake tool: 3.28.4
- Ninja Build: 1.12.0

I tried rebuilding and it showed a new error, this time it found the compiler but it was GNU (different from ARMClang). Error details are in the image below:

[image]

mcgordonite, 2024-05-28:
Is it possible the example you are using can target multiple compilers? It looks like the examples for K32L3A6 have different target types for each compiler. You can see this in the csolution.yml file:
```yaml
  target-types:
  - type: armgcc
    compiler: GCC
  - type: iar
    compiler: IAR
  - type: mdk
    compiler: AC6
```

You would need to select the mdk target type to build with AC6. That is selected in the Manage Solution dialog, opened from the cog icon at the top of the CMSIS view on the left hand side.

[image]

The compiler selected for your csolution needs to match what is in the vcpkg configuration.

NguyenTrongPhuc552003, 2024-05-29:
Based on your feedback, now I can use your extension more accurately and effectively for each type of project, board,... Reports and contributions will help the extension become more and more complete over time.

Thank you very much for your help!

closed 2024-05-29 by mcgordonite