ARM-software/vscode-keil-studio-pack #24: J-Link support & Manually install package & Configuration wizard display

Deadline039, 2024-07-26:
1. Currently, only ULINK, ST-Link and DAP are supported. I usually use J-Link for downloading and debugging. 
2. For some reason, github cannot be connected correctly in Chinese mainland. Can I manually install the CMSIS software package?
3. The configuration wizard display is a very useful tool for setting configuration items. Can this feature be added in future versions?

jkrech, 2024-08-05:
1. Note that you configure J-Link to run in CMSIS-DAP mode
2. Public CMSIS Software Packs can be download from https://www.keil.arm.com/packs/ or using cpackget from [CMSIS-Toolbox](https://www.keil.arm.com/artifacts/#tools/open-cmsis-pack/cmsis-toolbox) 
3. The configuration wizard view is already available in the current versions of Arm CMSIS Solution Extension. If you open a file with config wizard annotation: `//-------- <<< Use Configuration Wizard in Context Menu >>> --------------------` in VSCode
you can click on the "Open Preview" icon 
[image]
to open the graphical view.

Deadline039, 2024-08-06:
> 1. Note that you configure J-Link to run in CMSIS-DAP mode
> 2. Public CMSIS Software Packs can be download from https://www.keil.arm.com/packs/ or using cpackget from [...]

Thanks for your reply.

closed 2024-08-06 by Deadline039