ARM-software/vscode-keil-studio-pack #25: How to auto start the program after flashed?

jmjoy, 2024-08-22:
The task configuration:

```json
{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "arm-debugger.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "programs": "${command:cmsis-csolution.getBinaryFiles}",
            "cmsisPack": "Keil::STM32F1xx_DFP@2.4.1",
            "deviceName": "${command:cmsis-csolution.getDeviceName}",
            "processorName": "${command:cmsis-csolution.getProcessorName}",
            "problemMatcher": [],
            "label": "Flash Device",
            "connectMode": "underReset",
            "probe": "ST-Link"
        }
    ]
}
```

I am using the stm32f103c8t6 development board. 

Regardless of whether the connectionMode is `preReset` or `underReset`, the program cannot start automatically. I need to manually press the reset button of the development board or start debugger to start the program.

I don't know if this is a bug or unsupported, or if I have configured it wrong.

jkrech, 2024-08-26:
According to the documentation here https://developer.arm.com/documentation/108029/0000/Get-started-with-an-example-project/Run-the-solution-on-your-board?lang=en, I share your expectation that after the completion of the Flash Programming Algorithm, the connected target should be "reset" and the application should be started without the need to manually reset the device or start a debug session. The `preReset` and `underReset` only configure the reset handling during the debugger connection and should not impact the reset triggered after the debugger disconnects from the target.

Looking at the default task.json and your copy above I cannot see any difference other than the "connectMode" which you may want to remove.

jmjoy, 2024-08-26:
Maybe it's not a bug, it's just designed that way.

In the [arm-debugger documentation](https://developer.arm.com/documentation/108029/0000/Arm-Debugger-extension/Debug-your-project-with-Arm-Debugger/Arm-Debugger-debug-configuration-options?lang=en), there is an option `resetAfterConnect`, which can automatically reset, but the flash task does not have this option.

I am used to automatically reset the MCU after the burning is completed in stm32cubeIDE, so I have this question.

jreineckearm, 2024-08-26:
@jmjoy , thanks for reporting this!

@jkrech  is right that `preReset` and `underReset` connection modes are considered during the debugger connection. Not the disconnect. Hence, these would only have an impact if you set them on the launch config in a subsequent debug session. Same is true for `resetAfterConnect` which issues a reset after acquiring the debug connection but before loading any application file or handing over the control to the user. (This happens always for a flash download, hence is not configurable in the task configuration.)

The debugger should in fact always issue a reset after completion of the flash download. I can see this working with a different ST device. So, it could be something specific to this device...maybe about how the reset is wired in the chip. We'll try to reproduce with the exact same device.

In the meanwhile, could you please open a corresponding issue in https://github.com/ARM-software/vscode-arm-debugger and close this one here? As @jkrech indicated, this is a debugger issue which should be handled through that other repo.