ARM-software/vscode-keil-studio-pack #26: Provide manual commands to regenerate `CompileCommands` and `ClangdFile`?

jmjoy, 2024-08-26:
Type: Feature Request

I found that the `Arm CMSIS Solution` extension provides two options:
- `cmsis-csolution.autoGenerateCompileCommands` (*Automatically generate compile_commands.json and RTE files for the active csolution context*) 
- `cmsis-csolution.autoGenerateClangdFile` (*Automatically generate a .clangd file for the active csolution context*)

However, when significant project changes occur, clangd may fail, requiring a workspace restart to regenerate these files.

Is there a way to manually regenerate CompileCommands and ClangdFile?

jkrech, 2024-08-26:
When you open the "output panel" and select "CMSIS Build Manager" you should see that it invokes:
`cbuild setup <solutionname>.csolution.yml --context-set ...` whenever you e.g. edit and save the csolution.yml file this command is repeated. It will update the compile_commands.json file and copies it to the output artifacts folder.
You can manually invoke the command from the terminal window inside VSCode to trigger an update manually.
Does that help?

jmjoy, 2024-08-26:
Thanks.

However, I found that the `CompilationDatabase` configuration in the automatically generated `.clangd` file is: `/my-stm32/out/Project/STM32F103C8/Debug`

While the output of `cbuild setup` is: `[1/1] cd /my-stm32/tmp/1 && /home/jmjoy/.vcpkg/artifacts/2139c4c6/tools.kitware.cmake/3.28.4/bin/cmake -E copy_if_different /my-stm32/tmp/1/compile_commands.json /my-stm32/out/001-LED/STM32F103C8/Debug`.

The `CompilationDatabase` should be the directory containing `compile_commands.json`, but it doesn't match.

jmjoy, 2024-08-26:
Thank you very much.

Would it be more user-friendly to provide a VSCode command to manually run `cbuild setup`?

jkrech, 2024-08-26:
In my view, this should become part of `cmsis-csolution.refresh` which it currently does not seem to be.

closed 2024-08-26 by jmjoy