ARM-software/vscode-keil-studio-pack #28: clangd includes and macros not found in source code after running "Clean Output Directories" in CMSIS tab

j-silv, 2024-09-22:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 22.04
- VS Code Version: 1.92.2
- Extension Version: 1.18.3
- Target Device: NUCLEO-F446RE
- Other extensions you installed (and if the issue persists after disabling them): N/A
- A clear and concise description of what the bug is.

Whenever I open the workspace/project solution, there is no issue. clangd is able to parse all of the #includes and the headers. However, if I click on "Clean Output Directories" in the CMSIS solutions tab, then there are red squiggles for all of the included headers and symbols (clang(undeclared_var_use)). The issue goes away when I close the workspace and reopen it. I also notice that the clangd log shows "Failed to find compilation database".

**To Reproduce**
Steps to reproduce the behavior:
1. Create new `template` project
2. Verify clangd parsing works and definitions can be referenced from the source code
3. Run "Clean Output Directories"
4. See the red squiggles and inability to parse the header files

**Expected behavior**
When cleaning output directories, you lose access to the `compile_commands.json` file. However, when you rebuild the project and regenerate this file, clangd is still unable to parse the includes.

**Code sample and logs**

Pressing CTRL+S in the main.c source code file. clangd finds the include file definitions:
```
I[12:57:50.464] <-- textDocument/codeAction(8)
I[12:57:50.465] --> reply:textDocument/codeAction(8) 0 ms
I[12:57:50.465] --> textDocument/clangd.fileStatus
I[12:57:50.535] <-- textDocument/didSave
I[12:57:50.535] File version went from 1 to 1
I[12:57:50.535] --> textDocument/clangd.fileStatus
I[12:57:50.586] ASTWorker building file /home/user/files/lesson5/main.c version 1 with command 
[/home/user/files/tmp/1]
/home/user/.vcpkg/artifacts/2139c4c6/compilers.arm.armclang/6.22.0/bin/armclang --target=arm-arm-none-eabi -DSTM32F446xx -D_RTE_ -I/home/user/files/lesson5/RTE/_Debug_STM32F446RETx -I/home/user/.cache/arm/packs/ARM/CMSIS/5.9.0/CMSIS/Core/Include -I/home/user/.cache/arm/packs/Keil/STM32F4xx_DFP/2.17.1/Drivers/CMSIS/Device/ST/STM32F4xx/Include -isystem /home/user/.vcpkg/artifacts/2139c4c6/compilers.arm.armclang/6.22.0/include --target=arm-arm-none-eabi -mcpu=Cortex-M4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --target=arm-arm-none-eabi -mcpu=Cortex-M4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --target=arm-arm-none-eabi -c -mlittle-endian -std=gnu11 -Wno-macro-redefined -Wno-pragma-pack -Wno-parentheses-equality -Wno-license-management -g -O0 -o CMakeFiles/Group_Source_Files.dir/home/user/files/lesson5/main.o -c -D_ILP32=1 -D_USE_STATIC_INLINE=1 -D__APCS_32__=1 -D__ARMCC_VERSION=6220000 -D__ARMCOMPILER_VERSION=6220000 -D__ARMEL__=1 -D__ARM_32BIT_STATE=1 -D__ARM_ACLE=200 -D__ARM_ARCH=7 -D__ARM_ARCH_7EM__=1 -D__ARM_ARCH_EXT_IDIV__=1 -D__ARM_ARCH_ISA_THUMB=2 -D__ARM_ARCH_PROFILE='M' -D__ARM_EABI__=1 -D__ARM_FEATURE_CLZ=1 -D__ARM_FEATURE_COPROC=0xf -D__ARM_FEATURE_DSP=1 -D__ARM_FEATURE_FMA=1 -D__ARM_FEATURE_IDIV=1 -D__ARM_FEATURE_LDREX=0x7 -D__ARM_FEATURE_QBIT=1 -D__ARM_FEATURE_SAT=1 -D__ARM_FEATURE_SIMD32=1 -D__ARM_FEATURE_UNALIGNED=1 -D__ARM_FP=0x6 -D__ARM_FP16_ARGS=1 -D__ARM_FP16_FORMAT_IEEE=1 -D__ARM_NO_IMAGINARY_TYPE=1 -D__ARM_PCS=1 -D__ARM_PCS_VFP=1 -D__ARM_PROMISE=__builtin_assume -D__ARM_SIZEOF_MINIMAL_ENUM=4 -D__ARM_SIZEOF_WCHAR_T=4 -D__ARM_VFPV2__=1 -D__ARM_VFPV3__=1 -D__ARM_VFPV4__=1 -D__ATOMIC_ACQUIRE=2 -D__ATOMIC_ACQ_REL=4 -D__ATOMIC_CONSUME=1 -D__ATOMIC_RELAXED=0 -D__ATOMIC_RELEASE=3 -D__ATOMIC_SEQ_CST=5 -D__BIGGEST_ALIGNMENT__=8 -D__BITINT_MAXWIDTH__=128 -D__BOOL_WIDTH__=8 -D__BYTE_ORDER__=__ORDER_LITTLE_ENDIAN__ "-D__CHAR16_TYPE__=unsigned short" "-D__CHAR32_TYPE__=unsigned int" -D__CHAR_BIT__=8 -D__CHAR_UNSIGNED__=1 -D__CLANG_ATOMIC_BOOL_LOCK_FREE=2 -D__CLANG_ATOMIC_CHAR16_T_LOCK_FREE=2 -D__CLANG_ATOMIC_CHAR32_T_LOCK_FREE=2 -D__CLANG_ATOMIC_CHAR_LOCK_FREE=2 -D__CLANG_ATOMIC_INT_LOCK_FREE=2 -D__CLANG_ATOMIC_LLONG_LOCK_FREE=1 -D__CLANG_ATOMIC_LONG_LOCK_FREE=2 -D__CLANG_ATOMIC_POINTER_LOCK_FREE=2 -D__CLANG_ATOMIC_SHORT_LOCK_FREE=2 -D__CLANG_ATOMIC_WCHAR_T_LOCK_FREE=2 -D__CONSTANT_CFSTRINGS__=1 -D__DBL_DECIMAL_DIG__=17 -D__DBL_DENORM_MIN__=4.9406564584124654e-324 -D__DBL_DIG__=15 -D__DBL_EPSILON__=2.2204460492503131e-16 -D__DBL_HAS_DENORM__=1 -D__DBL_HAS_INFINITY__=1 -D__DBL_HAS_QUIET_NAN__=1 -D__DBL_MANT_DIG__=53 -D__DBL_MAX_10_EXP__=308 -D__DBL_MAX_EXP__=1024 -D__DBL_MAX__=1.7976931348623157e+308 -D__DBL_MIN_10_EXP__=(-307) -D__DBL_MIN_EXP__=(-1021) -D__DBL_MIN__=2.2250738585072014e-308 -D__DECIMAL_DIG__=__LDBL_DECIMAL_DIG__ -D__ELF__=1 -D__ESCAPE__= -D__FINITE_MATH_ONLY__=1 -D__FLT16_DECIMAL_DIG__=5 -D__FLT16_DENORM_MIN__=5.9604644775390625e-8F16 -D__FLT16_DIG__=3 -D__FLT16_EPSILON__=9.765625e-4F16 -D__FLT16_HAS_DENORM__=1 -D__FLT16_HAS_INFINITY__=1 -D__FLT16_HAS_QUIET_NAN__=1 -D__FLT16_MANT_DIG__=11 -D__FLT16_MAX_10_EXP__=4 -D__FLT16_MAX_EXP__=16 -D__FLT16_MAX__=6.5504e+4F16 -D__FLT16_MIN_10_EXP__=(-4) -D__FLT16_MIN_EXP__=(-13) -D__FLT16_MIN__=6.103515625e-5F16 -D__FLT_DECIMAL_DIG__=9 -D__FLT_DENORM_MIN__=1.40129846e-45F -D__FLT_DIG__=6 -D__FLT_EPSILON__=1.19209290e-7F -D__FLT_HAS_DENORM__=1 -D__FLT_HAS_INFINITY__=1 -D__FLT_HAS_QUIET_NAN__=1 -D__FLT_MANT_DIG__=24 -D__FLT_MAX_10_EXP__=38 -D__FLT_MAX_EXP__=128 -D__FLT_MAX__=3.40282347e+38F -D__FLT_MIN_10_EXP__=(-37) -D__FLT_MIN_EXP__=(-125) -D__FLT_MIN__=1.17549435e-38F -D__FLT_RADIX__=2 -D__FPCLASS_NEGINF=0x0004 -D__FPCLASS_NEGNORMAL=0x0008 -D__FPCLASS_NEGSUBNORMAL=0x0010 -D__FPCLASS_NEGZERO=0x0020 -D__FPCLASS_POSINF=0x0200 -D__FPCLASS_POSNORMAL=0x0100 -D__FPCLASS_POSSUBNORMAL=0x0080 -D__FPCLASS_POSZERO=0x0040 -D__FPCLASS_QNAN=0x0002 -D__FPCLASS_SNAN=0x0001 -D__GCC_ATOMIC_BOOL_LOCK_FREE=2 -D__GCC_ATOMIC_CHAR16_T_LOCK_FREE=2 -D__GCC_ATOMIC_CHAR32_T_LOCK_FREE=2 -D__GCC_ATOMIC_CHAR_LOCK_FREE=2 -D__GCC_ATOMIC_INT_LOCK_FREE=2 -D__GCC_ATOMIC_LLONG_LOCK_FREE=1 -D__GCC_ATOMIC_LONG_LOCK_FREE=2 -D__GCC_ATOMIC_POINTER_LOCK_FREE=2 -D__GCC_ATOMIC_SHORT_LOCK_FREE=2 -D__GCC_ATOMIC_TEST_AND_SET_TRUEVAL=1 -D__GCC_ATOMIC_WCHAR_T_LOCK_FREE=2 -D__GCC_HAVE_DWARF2_CFI_ASM=1 -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_1=1 -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_2=1 -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_4=1 -D__GCC_HAVE_SYNC_COMPARE_AND_SWAP_8=1 -D__GNUC_MINOR__=2 -D__GNUC_PATCHLEVEL__=1 -D__GNUC_STDC_INLINE__=1 -D__GNUC__=4 -D__GXX_ABI_VERSION=1002 -D__ILP32__=1 -D__INT16_C_SUFFIX__= "-D__INT16_FMTd__=\"hd\"" "-D__INT16_FMTi__=\"hi\"" -D__INT16_MAX__=32767 -D__INT16_TYPE__=short -D__INT32_C_SUFFIX__= "-D__INT32_FMTd__=\"d\"" "-D__INT32_FMTi__=\"i\"" -D__INT32_MAX__=2147483647 -D__INT32_TYPE__=int -D__INT64_C_SUFFIX__=LL "-D__INT64_FMTd__=\"lld\"" "-D__INT64_FMTi__=\"lli\"" -D__INT64_MAX__=9223372036854775807LL "-D__INT64_TYPE__=long long int" -D__INT8_C_SUFFIX__= "-D__INT8_FMTd__=\"hhd\"" "-D__INT8_FMTi__=\"hhi\"" -D__INT8_MAX__=127 "-D__INT8_TYPE__=signed char" -D__INTMAX_C_SUFFIX__=LL "-D__INTMAX_FMTd__=\"lld\"" "-D__INTMAX_FMTi__=\"lli\"" -D__INTMAX_MAX__=9223372036854775807LL "-D__INTMAX_TYPE__=long long int" -D__INTMAX_WIDTH__=64 "-D__INTPTR_FMTd__=\"d\"" "-D__INTPTR_FMTi__=\"i\"" -D__INTPTR_MAX__=2147483647 -D__INTPTR_TYPE__=int -D__INTPTR_WIDTH__=32 "-D__INT_FAST16_FMTd__=\"hd\"" "-D__INT_FAST16_FMTi__=\"hi\"" -D__INT_FAST16_MAX__=32767 -D__INT_FAST16_TYPE__=short -D__INT_FAST16_WIDTH__=16 "-D__INT_FAST32_FMTd__=\"d\"" "-D__INT_FAST32_FMTi__=\"i\"" -D__INT_FAST32_MAX__=2147483647 -D__INT_FAST32_TYPE__=int -D__INT_FAST32_WIDTH__=32 "-D__INT_FAST64_FMTd__=\"lld\"" "-D__INT_FAST64_FMTi__=\"lli\"" -D__INT_FAST64_MAX__=9223372036854775807LL "-D__INT_FAST64_TYPE__=long long int" -D__INT_FAST64_WIDTH__=64 "-D__INT_FAST8_FMTd__=\"hhd\"" "-D__INT_FAST8_FMTi__=\"hhi\"" -D__INT_FAST8_MAX__=127 "-D__INT_FAST8_TYPE__=signed char" -D__INT_FAST8_WIDTH__=8 "-D__INT_LEAST16_FMTd__=\"hd\"" "-D__INT_LEAST16_FMTi__=\"hi\"" -D__INT_LEAST16_MAX__=32767 -D__INT_LEAST16_TYPE__=short -D__INT_LEAST16_WIDTH__=16 "-D__INT_LEAST32_FMTd__=\"d\"" "-D__INT_LEAST32_FMTi__=\"i\"" -D__INT_LEAST32_MAX__=2147483647 -D__INT_LEAST32_TYPE__=int -D__INT_LEAST32_WIDTH__=32 "-D__INT_LEAST64_FMTd__=\"lld\"" "-D__INT_LEAST64_FMTi__=\"lli\"" -D__INT_LEAST64_MAX__=9223372036854775807LL "-D__INT_LEAST64_TYPE__=long long int" -D__INT_LEAST64_WIDTH__=64 "-D__INT_LEAST8_FMTd__=\"hhd\"" "-D__INT_LEAST8_FMTi__=\"hhi\"" -D__INT_LEAST8_MAX__=127 "-D__INT_LEAST8_TYPE__=signed char" -D__INT_LEAST8_WIDTH__=8 -D__INT_MAX__=2147483647 -D__INT_WIDTH__=32 -D__I__=1.0if -D__LDBL_DECIMAL_DIG__=17 -D__LDBL_DENORM_MIN__=4.9406564584124654e-324L -D__LDB [... cut ...] ain.c
I[13:06:05.765] --> textDocument/clangd.fileStatus
I[13:06:05.765] --> textDocument/clangd.fileStatus
I[13:06:05.765] --> textDocument/clangd.fileStatus
I[13:06:05.766] --> textDocument/clangd.fileStatus
```

**Screenshots**

Clean reopen of VS code workspace; macros can be parsed

[image]

Clicking on "Clean Output Directories" and then "CTRL+S" in main.c results in the following. The definitions are still not parsed even after "building" the solution.

[image]

**Additional context**
N/A

j-silv, 2024-09-22:
I think that the root issue is that the `compile_commands.json` is not being regenerated in the `out/` folder after it is deleted (either by the Clean Output Directories command, or by simply deleting the `out/` folder. I verified this... when the solution is reopened, then the `compile_commands.json` is regenerated in the out folder, but it is not regenerated after subsequent builds/rebuilds.

jkrech, 2024-09-23:
This sounds somewhat familiar. Which version of the CMSIS-Toolbox are you using?
Note that CMSIS-Toolbox 2.6.0 has been released a couple of weeks ago.
See issue: [#158](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/issues/158)

j-silv, 2024-09-29:
> This sounds somewhat familiar. Which version of the CMSIS-Toolbox are you using? Note that CMSIS-Toolbox 2.6.0 has been released a couple of weeks ago. @j-silv this should be fixed in version 2.6.0 of the CMSIS-Toolbox. See issue: [#158](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/issues/158)

I've tried with version 2.6.0 of the CMSIS-Toolbox but this issue is still present.

jkrech, 2024-09-30:
Can you please check whether the compile_commands.json is regenerated when you edit and save the active  csolution.yml file? 
Can you also please check the extension settings of the "Arm CMSIS Solution".
"Cmsis-csolution: Generate Clang Setup" needs to be enabled
You should be able to see the corresponding cbuild invokation in "Output" tab when choosing the drop down "CMSIS Build Manager".

[image]